dfl-rlab/dddmr_navigation #6: Question about changing lidar type.

moyahan, 2025-07-15:
First of all, thank you for your sharing. I noticed that in another issue, you mentioned that the current lego loam mapping and localization method supports repetitive scan lidar. Unfortunately, I am currently using the Livox Mid360 model lidar. May I ask if support for this model is in your future update plans? If not, I am considering either modifying the localization method to another approach, or adjusting how lego loam processes lidar data. Which method do you think would have the least impact on the system, and are there any other considerations I should be aware of? I look forward to your reply. Thank you!

tsengapola, 2025-07-17:
Thank you very much for your support about the bag collection.

Not much limitation for the environment, either indoor or outdoor should be fine.
We need the point cloud topic and if there is odometry topic should be even better.
About the bag file, we might need the lidar to be recorded statically around 10 seconds and then move straight couple meters should be enough for us to start.
Also we need the detail spec of lidar, i.e., vertical resolution, horizontal resolution, vertical fov.

tsengapola, 2025-07-29:
https://github.com/user-attachments/assets/eb1dc10c-a63a-4d48-a777-bd2333920b3e

@Xlqmu @moyahan We have upload a quick implementation for livox mid360, give it a try!
The known issue is that if you move too fast, the laser odometry might fail.
Please refer to branch:
https://github.com/dfl-rlab/dddmr_navigation/tree/livox_mid360

moyahan, 2025-07-29:
@tsengapola Excellent！I‘ll test it ASAP.

AstroBoy362, 2025-10-20:
@tsengapola Thank you very much！