dfl-rlab/dddmr_navigation #8: global path planner failed！！！

adl-123, 2025-08-15:
When I am performing global path planning, I often encounter the problem that the robot cannot accurately reach the target point nearby. The rotation controller keeps working, or DWA is blocked and cannot be stopped by pressing ctrl+c. Has the author encountered this problem before? @tsengapola

tsengapola, 2025-08-15:
It means there is obstacle in the inscribed_radius, so the path planning rejects your path and controller refuse to move the robot. 
The default value in the config is 0.5 meter.

https://github.com/dfl-rlab/dddmr_navigation/blob/64c625e6a3ea802c091734243d9ce1426bade5c8/src/dddmr_p2p_move_base/config/p2p_move_base_localization.yaml#L254

https://github.com/dfl-rlab/dddmr_navigation/blob/64c625e6a3ea802c091734243d9ce1426bade5c8/src/dddmr_p2p_move_base/config/p2p_move_base_localization.yaml#L291

If your robot is 0.6 meter long, then you can set inscribed_radius to 0.3 or a little bit larger say 0.32.

For ctrl+c issue, I think the log was printing so fast that it became difficult to kill. I will fix it by throttling the log.
Thank you very much for reporting this issue!!!

adl-123, 2025-08-15:
Thank you for your answer. Now I have another question. I have noticed that when I perform multiple planning operations, the planning speed becomes slower and slower, and sometimes no path can be planned. It seems that the planning has already succeeded, but the planning process hasn't ended yet, and the log is still being output. What could be the problem?

adl-123, 2025-08-15:
https://youtu.be/Ee0L__NJibs

adl-123, 2025-08-20:
@tsengapola Your modification is excellent. It seems that the problem has been solved. I will continue to use it in the future. Thank you!

[image]

adl-123, 2025-08-20:
@tsengapola I found another issue. I added a lidar as a sensor, but sometimes the global planning directly passes through obstacles without generating obstacle avoidance

[image]

video:
https://youtu.be/SXTjPLyy81k

adl-123, 2025-08-20:
https://youtu.be/09ia_vKVB88
This way, the problem becomes more apparent

adl-123, 2025-08-21:
@tsengapola 
Watch this video：https://youtu.be/I-t-78cR94M

adl-123, 2025-08-21:
@tsengapola
Watch this video：
https://github.com/user-attachments/assets/998a9a8b-05d1-4263-b570-9114d73f06f6

tsengapola, 2025-08-21:
Thank you for the video, it looks strange because even potential does not work.
I need couple days to write some tests to check.

adl-123, 2025-08-22:
@tsengapola  There are still such issues
[image]

adl-123, 2025-08-22:
@tsengapola OK, maybe replacing "lego loam" with "lio sam" might not be appropriate, but I don't think it should cause any issues with path planning. Could you help me analyze the root cause of this problem first?After modifying get_plan and the parameters, the problem still persists.

https://youtu.be/XnvSmS1a8_4

[image]
[image]

tsengapola, 2025-08-24:
I identified and can reproduce the issue.

Basically the issue is that when A* doing the expanding, we do not check line-of-sight because we want to save computation time. However, when expanding radius>2*inscribed_radius, this issue will occur.

https://github.com/dfl-rlab/dddmr_navigation/blob/8d102bc3782d5305472f052b945ceebf6e9e3269/src/dddmr_global_planner/src/a_star_on_pc.cpp#L198

If you change above line from 1.0 to 0.5, I think the path will be correct. Because 0.5<2*your_inscrebed_radius. Make sure your ground voxel size is smaller than 0.5 m.

But reducing search radius is not a good solution. We are working on an better solution that we will check line-of-sight when expanding. In addition, I will parameterize this value for user to setup. The new release will require couple days.

adl-123, 2025-08-25:
You are right, the problem indeed lies here. Looking forward to your new release, thank you for your efforts

adl-123, 2025-08-28:
https://youtu.be/fQIXKv6Gmfw
Hello, I tried your latest code. In my case, is it because some parameters are not set correctly?
@tsengapola When I increase the perception_window_size in perception_3d_global, such a warning appears, causing the path planning to be delayed.

[image]

tsengapola, 2025-08-28:
Change look ahead distance to 10.0 meters enable the robot to plan earlier when seeing the wall. It should solve the issue.

https://github.com/dfl-rlab/dddmr_navigation/blob/eb691285a584c0a07cfac0ae0b52da4a84b21cb3/src/dddmr_p2p_move_base/config/p2p_move_base_localization.yaml#L338

Setting perception window size larger will include a lot of point cloud for marking and clearing, result in slow speed. You can set it to 5-10 meters, should be fine once look ahead become larger.
But I still would like to know what is your point cloud size:
```
ros2 topic bw non_ground_cloud
```

tsengapola, 2025-08-28:
It looks like look_ahead_distance is still 2.0 meters in your video. Because the only modify a short pah in the green path.
Do you see logs something like this:
```

[global_planner_node-2] [INFO] [1756372591.313455761] [dynamic_window_aware_global_planner]: look_ahead_distance: 100.00
[global_planner_node-2] [INFO] [1756372591.313558624] [dynamic_window_aware_global_planner]: recompute_frequency: 10.00

```

adl-123, 2025-08-28:
[image]

[image]
Do you mean we should change it to 10.0 here as well?
What I mean is, have you noticed that at the end of the video, the global path directly goes through the obstacle without any error, and then enters wait

adl-123, 2025-08-29:
I used `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`, applied the `issue8` branch, and changed `look_ahead_distance` to 10.0. However, this scenario still fails to pass
@tsengapola 
https://github.com/user-attachments/assets/e514d1ca-1041-4c2c-acf5-ab1c712aba38

adl-123, 2025-08-29:
https://github.com/user-attachments/assets/0209dbf8-9b54-4793-96c7-ee97bac4b3ba

tsengapola, 2025-08-29:
can you show me
```
git log
```

adl-123, 2025-08-29:
[image]

adl-123, 2025-08-29:
[20250827.zip](https://github.com/user-attachments/files/22037339/20250827.zip)

tsengapola, 2025-08-29:
Can you also provide the point cloud in video for around 30 seconds. Also what is your ground voxel size?

https://github.com/user-attachments/assets/6d237445-c6db-40a8-9eb9-6414bcc3172c

adl-123, 2025-08-29:
[image]

[dddmr.db3.zip](https://github.com/user-attachments/files/22039581/dddmr.db3.zip)

tf in rosbag,I've noticed a slight issue with my self-positioning. The z-value seems to be getting lower and lower. Will this affect the planning? I observed that the problematic wall doesn't appear as dark in color in dGraph

tsengapola, 2025-09-01:
[image]

In theory, when the position is correct, it means you are minimize the delta Z, so the value of dGraph is somehow the distance of xy plane.

Regarding the z value issue, the robot should get close to the correct ground so that it can deal with multi-floor problem:
From the picture, projection to the ground is based on the delta Z, if the z value is incorrect, the selection for the ground point will be wrong.

Regarding, the z removal for the dGraph value, I think it should be no harm to remove z, I will work on a new commit for it.